ntruchsess/arduino_uip #110: UIP lib on Arduino Uno+IE Shield (Itead Studio ENC28J60 shield)-Server unresponsive

rubborto, 2015-01-26:
Hi Norbert
and thanks for having developed this amazing library. I am not a big fan of the ENC ethernet chip, but I have dedicated it some time.
I'm facing some trouble with a very basic server sketch based on your example "TcpServer". After uploading the sketch, the first request sent from another PC's browser goes answered by the server very quickly. A second request posted sometime after (such as 20-30'' after the first), struggles to  come back (takes more than one minute and does not come back). After sometime (>1') however looks like the server responds again. 
I do not know why or what is preventing the server from ending properly the first request and set itself to receive the next one promptly. 
I suppose if I keep sending requests distanced between themselves by 15-20" (even shorter) the server should respond to all of them properly and timely. 
If I distance the requests by several minutes (e.g. more than 3-4'), the server responds properly.
I suspect there is some timer or variable that I could twist to improve this issue?
Many thanks in advance
Ruben

manosv, 2015-01-26:
I had many problems myself but after reading some tips from Norbert where
he recommends to NOT use any delays in the code and calling
ethernet.maintain() every 100ms the problems disappeared
. BTW i use the branch fix_erata12

rubborto, 2015-01-27:
Thanks manosv.  how do you place ethernet.maintain in the code such as to have a 100ms delay between each call? I suppose the loop code executes and takes its time do it according to the instructions. I have no idea how I can have ethernet.maintain executed every 100ms or so...if you put a delay then you contradict your first statement.
and then ethernet.maintain is a function required to maintain the ip address, I guess I'm not having this problem?
and what is "branch fix-erata12?"
thanks again

manosv, 2015-01-27:
Norbert proposed to have a timer interrupt every 100ms or so where you
execute theethernet.maintain() call .Take note that it doesnt need to be
exactly 100ms but no more than 100ms.This is the forum topic where he
suggests various things i suggest to read the whole topic

http://forum.arduino.cc/index.php?topic=178024.msg1612316#msg1612316