ntruchsess/arduino_uip #120: 2 UDP frames to decode

JBLSteeve, 2015-07-18:
I use Jeedom to send XPL message to my arduino.
We have the heardbeat message and just after the command message (see on the socket test capture)
[image]

With UIPEthernet i use the udpserver example to receive message but i receive just the first one (heartbeat). You can see the log
[image]

The strange think is some time i can receive sometime not. Can you help me ?

JBLSteeve, 2015-08-04:
Nobody can help me, i see some configuration in the library but i don't know what can i change

bholmes451, 2015-08-04:
I haven't seen any real activity in this project in the last 8 months I
have been monitoring it.  I don't have the technical skills right now to
make any changes myself.  I did get it working back then but haven't
touched it in the last 8 months.  I'm sure people would appreciate anyone
who can actually help on this project because it seems real useful but not
currently maintained...

Ben

JBLSteeve, 2015-08-04:
Thanks, for you feedback. I'm in the same state as you i have not the skill to understand and do some changes

rodjt, 2015-08-04:
Same here. I'm using the lib very successfully, but don't have the skills to make changes.
Please, someone who's got the experience, this is worth supporting.....

gibsop1, 2015-08-04:
I wonder if it is just a coincidence that approximately 10 months ago the
ESP8266 made it's appearance.  Shame as so much work went into this, but
it's to be expected when such a cheap, full featured chip comes on the
market - not to mention the fact that the ESP8266 is wifi.

-- 

Please consider the environment before printing this email
This e-mail and any attachment transmitted with it are confidential and are
intended for the named person’s use only. The contents of the e-mail may
contain sensitive and private or legally privileged information. If you
have received this e-mail in error, please delete it (and any attachment)
from your system and notify the sender immediately. You should not retain,
copy or print the contents of this e-mail (or any attachment) nor should
you, directly or indirectly, disclose or distribute the contents to anyone.
The opinions, advice or judgments within this email are not necessarily the
views shared by BlueIceCube.

BlueIceCube is a limited company registered in England and Wales.
Company Registration No. 8885577
Registered Offices: 46 Seacrest Avenue, Cullercoats, North Shields, NE30 3DP

rodjt, 2015-08-04:
Just checked out the ESP8266. Very nice little module and I guess in most cases wifi is available.
My application is 'in the field' and I only have an RJ-45 Ethernet connection available, so the ENC28J60 is a really good fit. Hopefully someone with the experience will still get on board.