ntruchsess/arduino_uip #125: Error while compiling for Arduino Due

simongerads, 2015-09-06:
I have tried to compile the TcpServer Example with an Arduino Due and the enc28j60 but I always run into the following error message. 
Not sure if this is an library issue but this is what I get:

Arduino: 1.6.5 (Mac OS X), Board: "Arduino Due (Programming Port)"

WARNING: library UIPEthernet claims to run on [avr] architecture(s) and may be incompatible with your current board which runs on [sam] architecture(s).

/Users/XXX/Documents/Arduino/libraries/arduino_uip-master/utility/Enc28J60Network.cpp:29:20: fatal error: avr/io.h: No such file or directory
 #include <avr/io.h>
                    ^
compilation terminated.
Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

CodeSurfing145, 2015-10-15:
I got the same error.
Haven't found a fix yet >_<

CodeSurfing145, 2015-10-28:
Using Releases > v1.59 it does compile and upload but doesn't start the Ethernet controller.