ntruchsess/arduino_uip #126: Build issue when activating IPv6 support

hekra01, 2015-09-17:
Hi

* I activate IPv6 support like this:
```
--- src/utility/uipethernet-conf.h.bk	2015-09-17 13:11:16.670537859 -0400
+++ src/utility/uipethernet-conf.h	2015-09-17 12:52:40.332102165 -0400
@@ -1,6 +1,7 @@
 #ifndef UIPETHERNET_CONF_H
 #define UIPETHERNET_CONF_H
 
+#define UIP_CONF_IPV6 1
 
 /* for TCP */
 #define UIP_SOCKET_NUMPACKETS    5
```
* Then build fails with
```
/usr/share/arduino/libraries/UIPEthernet/src/utility/uip.c:87:26: fatal error: uip-neighbor.h: No such file or directory
 #include "uip-neighbor.h"
                          ^
compilation terminated.
Error compiling.
```
* and indeed `uip-neighbor.h` does not exist

Am I missing something?

ntruchsess, 2015-10-02:
ipv6 support is not included, it worked out to be way to large for an Uno. It might run on a Mega256 or Due though as they have more memory. If you happen to proceed there including it (check the original µIP sourcecode for the missing files) I'd be happy to see a pull-request if you get it running.

closed 2015-10-02 by ntruchsess

nethope, 2016-11-21:
I couldn't get IPv6 working with this library either, but I got everything working just the way I wanted with [EtherSia](https://github.com/njh/EtherSia) on my Nano v3. That is a very lean network stack for Arduino, supporting ENC28J60, W5500, and W5100 Ethernet controllers; however it is IPv6-only (which is fine for me, maybe better).