ntruchsess/arduino_uip #144: Lockups after random time, able to receive but not send data

jeskata, 2016-01-26:
I have an Arduino Mega running both a web server and a UDP server according to the examples.

Web server is being read twice an hour for measurement data. UDP sent to unit only occasionally for home automation commands.

After a random amount of time (lastly around three days), like many others also have reported, it stops responding. According to debug prints, reception is ok and it attempts to write data back on the client, but nothing is sent. UIPServer::write and UDP::write both return zero when this happens.
I have tried running begin(MAC) and restarting servers when it happens but it does not seem to help.

Only a hard reset (have made a hardware watchdog to do this) works.

This could lead me thinking that it is a SPI related problem but I have another unit (CC110l radio module) on the bus that keeps working when this happens.

What does a hard reset reset of the AVR do to the ENC28J60 that begin(MAC) does not do? ENC28J60 reset pin not wired.

burner-, 2016-01-31:
You can reproduce that with flooding udp at ~100ms rate. With 100B packets it take only ~20seconds to reproduce this. With 10B packets it take ~minute or two.

burner-, 2016-01-31:
Actually looks that in my case write return right number of bytes even it fails. Anyway looks that Enc28J60.init(mac); fix problem if you know when to do it. In my case I make tcp server and poller. If tcp server does not get poll each 60 seconds it run Enc28J60.init(mac);

lcafaro, 2016-02-08:
Hello,

seems like an issue related with the well known ENC28J60 [Errata #12 and #13](http://ww1.microchip.com/downloads/en/DeviceDoc/80349b.pdf) bugs.

Have you tried using the code from branch `fix_errata12`? Download link: https://github.com/ntruchsess/arduino_uip/archive/fix_errata12.zip

I've noticed that the fix for Errata 12 has been merged into master, but still the `fix_errata12` branch has more controls about the transmitter hang, so I'd try compiling with that library version and check if it's stabler.

jeskata, 2016-02-08:
Have also encountered a different kind of lockup, once again today - in this case it does acknowledge incoming packet at all - no interrupt is given.

jeskata, 2016-02-10:
Took the Errata12 fix into use, lets see then...

jeskata, 2016-02-11:
Nope, it had rebooted on watchdog after around 24 hours uptime, after PC tried to read data.

lcafaro, 2016-02-11:
Strange, my code (built from the errata12 branch) has been up and running for 3 days now.

Could it be something else causing the issue? Like a memory leak somewhere...

jeskata, 2016-02-12:
I have tried checking free memory while this was running on a 1280, over 5k free at time of lockup.

lcafaro, 2016-02-12:
Could it be the other host not responding anymore for a moment? It may cause the connection to wait forever if your code is blocking (although it should recover after a timeout...).

Other than that I'm starting to run out of ideas...

jeskata, 2016-02-12:
I have another ENC28J60 that I could set up to be accessed similarly (or more often) to see if it happens with that too. Both the Arduino and the PC are on the same switch.