ntruchsess/arduino_uip #155: SAMD21 (Arduino Zero) Support

ljbeng, 2016-05-23:
Will this library be compatible with the SAMD21 any time soon?

Daniel-dk, 2016-10-19:
Ethercard uses direct Register access on the AVR, which won't work on ARM.

a simple ( or simplistic ) way of looking at it is to make sure reads, writes and chipselects are changed to use the SPI library instead of direct register usage ( which is specific to AVR only in this case ).

The SPI interface will have to be rewritten/replaced by the Arduino SPI library's interace
things like` SPDR = *data++;` can be replaced by `SPI.transfer(*data++);` for example.

- i don't have one of these at the moment, but am looking at it for a future project so might have to go down that rabbithole...

dimitar-kunchev, 2017-11-09:
I did get UIPEthernet working on SAMD21E16, but I started with https://github.com/UIPEthernet/UIPEthernet. Found all defines for ARCH_SAM and added checks for ARCH_SAMD next to them. So far testing for few days and looks OK-ish. There are glitches but they may be unrelated.