ntruchsess/arduino_uip #156: WebClient only works occasionally

HoffaSweden, 2016-05-25:
Hello

I'am trying to get an arduino webclient code to work. I am trying to fetch www.google.com

My issue is: It only works **occasionally**, lets say 1 time out of 5.
4 times out of 5 it times out and says "connection failed."

I am using an arduino UNO and ENC28J60 marked: ENC28J60-I/SO e3, 1246A7S
I am using arduino ide 1.5.8 and UIPethernet 1.59

What can be wrong? I have tried the following / I know the following:

-Level translator on 3,3V data pin from ENC28J60 to arduino. No difference at all.
-Turned on debugging and sees that ARP runs.

This is my sketch, and further below the response code when functioning and failing

#include <SPI.h>
#include <UIPEthernet.h>

EthernetClient client;

void setup() {

  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  byte mac[] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };
  char server[] = "www.google.com";
  IPAddress ip(192, 168, 77, 177);
  Ethernet.begin(mac);

  Serial.print("localIP: ");
  Serial.println(Ethernet.localIP());
  Serial.print("subnetMask: ");
  Serial.println(Ethernet.subnetMask());
  Serial.print("gatewayIP: ");
  Serial.println(Ethernet.gatewayIP());
  Serial.print("dnsServerIP: ");
  Serial.println(Ethernet.dnsServerIP());
 
  delay(1000);
  Serial.println("connecting...");

  // if you get a connection, report back via serial:
  if (client.connect(server, 80)) {
    Serial.println("connected");
    // Make a HTTP request:
    client.println("GET /search?q=arduino HTTP/1.1");
    client.println("Host: www.google.com");
    client.println("Connection: close");
    client.println();
  } else {
    // if you didn't get a connection to the server:
    Serial.println("connection failed");
  }
}

void loop() {
  // if there are incoming bytes available
  // from the server, read them and print them:
  if (client.available()) {
    char c = client.read();
    Serial.print(c);
  }
  
  //Serial.print("-"); 

  // if the server's disconnected, stop the client:
  if (!client.connected()) {
    Serial.println();
    Serial.println("disconnecting.");
    client.stop();

    // do nothing forevermore:
    while (true) {
       //Serial.print("EL"); 
    }
  }
}

***************** serial print out when functioning

localIP: 192.168.77.54
subnetMask: 255.255.255.0
gatewayIP: 192.168.77.1
dnsServerIP: 192.168.77.1
connecting...
connected
HTTP/1.1 302 Found
Location: http://www.google.se/search?q=arduino&gws_rd=cr&ei=Wa5FV9ipLMq6swHt9Lm4Cg
Cache-Control: private
Content-Type: text/html; charset=UTF-8
P3P: CP="This is not a P3P policy! See https://www.google.com/support/accounts/answer/151657?hl=en for more info."
Date: Wed, 25 May 2016 13:53:29 GMT
Server: gws
Content-Length: 278
X-XSS-Protection: 1; mode=block
X-Frame-Options: SAMEORIGIN
Set-Cookie: NID=79=H2CvpT9Y55kdPccpqfrPq5R6I3JyFXopiHP1jskstWS8mG8tOed0hHypMhuV2ww_--OdmSRGgREoNuozifunA9cPx07jL-H45UplTXJbaQFSVsbDSG59qb_3kQb6ePAy; expires=Thu, 24-Nov-2016 13:53:29 GMT; path=/; domain=.google.com; HttpOnly
Connection: close

<HTML><HEAD><meta http-equiv="content-type" content="text/html;charset=utf-8">
<TITLE>302 Moved</TITLE></HEAD><BODY>
<H1>302 Moved</H1>
The document has moved
<A HREF="http://www.google.se/search?q=arduino&amp;gws_rd=cr&amp;ei=Wa5FV9ipLMq6swHt9Lm4Cg">here</A>.
</BODY></HTML>

disconnecting.

*******************

********************serial print when not functioning

localIP: 192.168.77.54
subnetMask: 255.255.255.0
gatewayIP: 192.168.77.1
dnsServerIP: 192.168.77.1
connecting...
connection failed

disconnecting.

mfe5003, 2016-05-28:
I don't know the cause, but I can confirm that DHCP will fail for me about every other time I initialize a device, however the static configuration is reliable.

markfoodyburton, 2016-11-08:
I have exactly this issue.
Once it works - it works for a long time (several hundred requests), when it fails, it keeps failing.
In my case, when it fails, I reset the device, and the arduino, and try again. 
I am using DHCP, which works reliably for me - every time I get a valid IP address, however it fails to connect to the remote server.

Any ideas appreciated !