ntruchsess/arduino_uip #158: Low Memory when using Arduino Nano

ido-ran, 2016-06-27:
Hi,
I'm trying to use UIP for a project I'm building with Arduino Nano and this is the result of compiling a sketch that use only UIP, without any other code.

Sketch uses 22,126 bytes (72%) of program storage space. Maximum is 30,720 bytes.
Global variables use 1,545 bytes (75%) of dynamic memory, leaving **503** bytes for local variables. Maximum is 2,048 bytes.
**Low memory available, stability problems may occur.**

As you can see there are only 503 bytes left and it is considered low memory.
I'm interested to know if there is a way to remove parts that I don't need from UIP such as the server and UDP classes?

Thank you,
Ido.

mfe5003, 2016-06-28:
There is a configuration setting in utility/uipethernet-conf.h that can turn off UDP if you dont need it.

ido-ran, 2016-06-28:
Thank you. Is there a way to turn it off from the sketch without changing the source code?

mfe5003, 2016-06-28:
Not within the arduino IDE.  
You could try compiling the library with a different output file name and the correct value of the macro passed, then link to the new object when building the hex file.

I'm don't know if that is the correct way to do it, I would just change the source.

ido-ran, 2016-06-28:
Ok, thanks

ido-ran, 2016-06-30:
I've try set `UIP_CONF_UDP` to `0`
and I got this error:
`MyProject.ino:111: undefined reference to `UIPEthernetClass::begin(unsigned char const*)'`

If I change it back to 1 it compile just fine.

mfe5003, 2016-06-30:
Without seeing your code, I am guessing that you're trying to use DHCP.
That protocol requires UDP.
Try compiling with a static address.

ido-ran, 2016-06-30:
Yes, you are right :)

ido-ran, 2016-07-01:
New issue (still related to size limitation). I've actually run out of memory, I have 27 bytes left and it cause the Arudino to restart when calling Ethernet.begin function.
So, I've change `UIP_CONF_UDP` to `0` and set all the begin parameters hardcoded exactly like it was getting from the DHCP.
The problem is now the connection to the server is always failing. I've try different IPs but they all fail.

Are you sure there is no other things I can shave off, as DHCP is very useful.