ntruchsess/arduino_uip #162: Not able to receive more than 1500 bytes

dimitre, 2016-08-29:
I was investigating an issue while receiving large blobs from OSC
and it seems to be a limit defined in UIPUdp.h
UIP_UDP_MAXDATALEN
but even editing this value the packet seems to stop looping in about less than 1500 bytes.
Thank you

OKNoah, 2017-01-11:
I may be having the same issue, but it's closer to around 300 bytes.

dimitre, 2017-01-11:
@OKNoah I'm not sure because I dont know much about networking, but I've found a hard limit in my router of  1500 bytes per packet.

csBlueChip, 2017-07-19:
1500 is the "defacto norm" for a network MTU ...So I strongly suspect the 1500 byte limit will be the maximum size your network can handle ...And I suggest you leave it alone unless you know what you're doing, and what horrific things might happen if you change it.