ntruchsess/arduino_uip #163: Library conflict with interrupt routine or pins

Paul-Ver, 2016-09-21:
I must say you've got quite a nice library, however, it seems to break my code for no apparent reason.

I've got a encoder (Keyes K-040) with built-in button, and would like to make it an "ethernet dimmer".
My code works very well actually, but when I use "Ethernet.begin(mac)", it just wont register encoder changes anymore.
Interesting fact is, that the button works regardless of the Ethernet being active or not.

I've attached my code below and will check if it works without interrupts.
(But I don't see why it should be a problem)

`
#include <UIPEthernet.h>

volatile int value = 0;
volatile bool up;
int previousValue = 0;

const int PinCLK = 2;                 // Used for generating interrupts using CLK signal
const int PinDT = 3;                  // Used for reading DT signal
const int PinSW = 4;                  // Used for the push button switch

void isr ()  {                    // Interrupt service routine is executed when a HIGH to LOW transition is detected on CLK
  up = (digitalRead(PinCLK) == digitalRead(PinDT));
  if (up) {
    value++;
  } else {
    value--;
  }
}

void setupEthernet(){
  uint8_t mac[6] = {0x00, 0x01, 0x02, 0x03, 0x04, 0x05};
  //Ethernet.begin(mac); // When uncommented, code breaks for no apperant reason.
  Serial.print(F("localIP: "));
  Serial.println(Ethernet.localIP());
  Serial.print(F("subnetMask: "));
  Serial.println(Ethernet.subnetMask());
  Serial.print(F("gatewayIP: "));
  Serial.println(Ethernet.gatewayIP());
  Serial.print(F("dnsServerIP: "));
  Serial.println(Ethernet.dnsServerIP());
}
void setupPins(){
  pinMode(PinCLK, INPUT);
  pinMode(PinDT, INPUT);
  pinMode(PinSW, INPUT_PULLUP);
  interrupts();
  attachInterrupt (0, isr, CHANGE); // interrupt 0 is always connected to pin 2 on Arduino UNO  
}

void setup ()  {
  Serial.begin (115200);

  setupEthernet();
  setupPins();
  
  Serial.println(F("Start"));
}

void loop(){

  if (!(digitalRead(PinSW))) {      // check if pushbutton is pressed
    Serial.println(F("Button press!"));
    delay(1000);//Only do this once a second, to avoid spamming the network/or toggling fast.
  }

  if (previousValue != value)  {        // do this only if rotation was detected
    Serial.print (F("Diff = "));
    Serial.println (value-previousValue);
    previousValue = value;
    delay(100);
  }

}
`

Paul-Ver, 2016-09-21:
I'm now using interrupt pin 3 (interrupt source 1) instead of pin 2 (interrupt source 0)
And I use pin 4 (instead of pin 2) for the other regular I/O pin.

This seems to work well with the library.
If you see anything logical, please share.
If you think it's worth investigating (or have any ideas to try), please say so.
If you don't bother, you can close this issue.

It may be something with the power, since it's on Arduino Nano with Nano Ethernetboard.
I'm powering it through USB, so that could be a problem.
But it really doesn't make sense why using other pins would change anything.

I was also questioning that it could have something to do with my board, but that seems illogical since it does work without the Ethernet library started.

cmoine, 2018-06-06:
Arg, same problem here, and I need both hardware interrupt: 2 & 3 :( Any idea where pin 2 it is used in the library ?

Bjarne-J, 2020-07-23:
If it's still relevant to someone:

I've had the same experience recently by using this combination:

- Arduino Nano V3.0
- Nano Ethernets Shield V1.0
- Library UIPEthernet

In my application I was using both interrupts on pin D2 and D3, respectively.

When activating **_Ethernet.begin(mac, ip)_** the interrupts ceased to work.  In my application I was able to reduce the interrupt to pin D3 and separate pin D2 from the circuit. Then everything was ok.

It seems that the **_UIPEthernet_** library uses the D2 pin for some internal purpose.

If one still requires two interrupts on the Nano when having the Ethernet shield attached, it might be possible to use this approach: https://www.brainy-bits.com/make-any-arduino-pin-an-interrupt-pin/

However, I have not tried this.