ntruchsess/arduino_uip #168: UDP receive works only once

thecpcat, 2016-11-27:
Hello!
I'm novice and using [Agentuino](https://github.com/carlosdelfino/agentuino) library to make an SNMP agent. After reset it responds correctly one time but then stops. If I understand debug correctly, the snmp library stops receiving udp packets from the driver, udp parsePacket is not happening. Arduino is responding to pings OK all the time. What may be the problem here? 

The debug log with one successfull:

> receivePacket [F4-147], next: 14C, stat: C0, count: 1 -> OK
receivePacket: 255
readPacket type IP, uip_len: 83
udp, uip_newdata received packet: 1, size: 41
freeing packet: 255
udp parsePacket received packet: 1, size: 41
UDP Packet Received Start.. RAM:6185
OID: 1.3.6.1.2.1.1.1.0
sysDescr...SNMPtemp 8
Sending response...
udp beginPacket, rip: 192.168.1.50, port: 65214
udp, uip_poll preparing packet to send: 1, size: 99
udp, uip_packet to send: 1
Enc28J60Network_send uip_packet: 1, hdrlen: 42
sendPacket(1) [800-863]: 0 AC 22 B 26 94 94 DE AD BE EF FE ED 8 0 45 0 0 55 0 1 0 0 40 11 F6 49 C0 A8 1 CB C0 A8 1 32 0 A1 FE BE 0 41 C5 3F 30 37 2 1 0 4 6 70 75 62 6C 69 63 A2 2A 2 4 0 0 43 4A 2 4 0 0 0 0 2 4 0 0 0 0 30 16 30 14 6 8 2B 6 1 2 1 1 1 0 4 8 53 4E 4D 50 74 65 6D 70 
freeing PDU.. RAM:6185
UDP Packet Received End.. RAM:6185

And one bad attempt:
> receivePacket [250-2A3], next: 2A8, stat: C0, count: 1 -> OK
receivePacket: 255
readPacket type IP, uip_len: 83
freeing packet: 255

thecpcat, 2016-12-10:
Workaround via adding udp stop+start to the Agentuino library code.

closed 2016-12-10 by thecpcat

anthonyn, 2017-08-03:
iI,

I think I am seeing a similar thing, can you share any more info on your solution?

Thanks,

thecpcat, 2017-08-22:
I've added:
```
Udp.stop();
Udp.begin(SNMP_DEFAULT_PORT);
```
After each request processed by the SNMP library. It's probably a dirty hack, but worked for me. Don't remember the details though.