ntruchsess/arduino_uip #169: Library won't work when gateway sets up

airween, 2016-12-01:
Hi there,

I'ld like to build a simple remote switch through HTTP. The examples are good points to start.
But there is a problem: when I set up the gateway, the program doesn't work. Without it, the code works as well.

```c
IPAddress ip(192, 168, 72, 3);
IPAddress gateway(192,168,72,254);
IPAddress subnet(255,255,255,0);
EthernetServer server(80);

void setup() {
  Serial.begin(9600);
  Ethernet.begin(mac, ip, gateway, gateway, subnet);
```
This dosen't work. If I left the last 3 arguments, then HTTP server works as well.

Could anybody helps me?

Regards,
a.

airween, 2016-12-01:
Well, I've removed all Serial function call (no Serial.begin(), any .print()...), and the problem solved.
Is there any explanation for this?

stevstrong, 2016-12-02:
Which hardware/board are you using?
Which software?

airween, 2016-12-02:
Hi stevstrong,

I'm using arduino nano with ENC28J60.  Library is uipethernet.

Thanks,
a.

stevstrong, 2016-12-02:
@airween 
The Print class for AVR may have a [problem with the "F()" macro](https://github.com/arduino/Arduino/blob/2bfe164b9a5835e8cb6e194b928538a9093be333/hardware/arduino/avr/cores/arduino/Print.cpp#L114), as also did the [STM32duino](https://github.com/rogerclarkmelbourne/Arduino_STM32/issues/231), where I am involved in.
For the Ethernet.begin it would be normal to use only the MAC address, since the IP number is normally  automatically allocated by the router (DHCP).

airween, 2016-12-02:
Hi @stevstrong, thanks for your reply.
That's a bad news for me - now I don't know, what can I do.
Anyway, meantime I've ran into a (several) new problems. Eg., I've removed all Serial.*() call from code, but the explicit IP settings doesn't work again... I'll check the DHCP settings, but... it would be better to use it as explicit. Another example: when I change the TCP port (from 80 to another one, eg. 81), then sometimes it doesn't work again... but sometimes it start to work... It drives me crazy....

airween, 2016-12-03:
Another issue - that's not linked to this original issue. Here is a small "syslog" routine:
```c
void syslog(char *msg) {
    udp.beginPacket(IPAddress(192,168,72,254),514);
    udp.write("IPBridge ");
    udp.write(msg);
    udp.endPacket();
    udp.flush();
    udp.stop();
}
```
and here is why do I use:
```c
  if (client)
  { 
    //Serial.println("-> New Connection");
    syslog("New connection");
```
There is a cron script, which checks this board that responds or not in every 10 minutes. On syslog server, I see the results:
```text
Dec  2 18:17:53 IPBridge New connection
Dec  2 18:20:01 IPBridge New connection
Dec  2 18:20:02 IPBridge New connection
...nothing...
Dec  3 08:50:01 IPBridge New connection
Dec  3 08:50:01 IPBridge New connection
Dec  3 09:00:01 IPBridge New connection
Dec  3 09:00:01 IPBridge New connection
```
While the elapsed time there isn't any syslog message from board - but the board has worked as well, I'm logging the result of cron script, and there seems that board is working. But the sending of UDP packets doesn't work randomly.

Do you have any idea, why does it happen?

Thanks,
a.

stevstrong, 2016-12-03:
Ervin, sorry I am not that deeply involved in UIP, but I remember I also had several issue with my ENC28J60 module and all associated software (Ethercard as well), that's why I moved to a W5500 module.
It may cost a bit more ([around $5 on Aliexpress](https://www.aliexpress.com/item/W5500-Ethernet-network-module-hardware-TCP-IP-51-STM32-microcontroller-program-over-W5100/32776847766.html)), but is working **RELIABLE** and much faster and you can save a lot of time avoiding these kind of issues :).

airween, 2016-12-03:
He @stevstrong - thanks again your feedback.
Yep, you're right - the board has run about 30 hours ago, now it answers all requests, but the syslog doesn't work much. It sometimes sends the message, sometimes not... That's absolutely unpredictable.
I'll see your suggestion - thanks.

Regards,
a.

stevstrong, 2016-12-03:
No problem (szivesen :))
As alternative, you could save some cents by [buying a W5100 module](https://www.aliexpress.com/item/TOP-Mini-W5100-LAN-Ethernet-Shield-Network-Module-board-Best/32561025589.html).
Advantage of WIZ modules vs. ENC28: lower flash and RAM usage, full Arduino compatibility.

airween, 2016-12-03:
Wow! :)
Thank you (köszönöm :)) again. I've found some other modules in .hu, in a chinese webshop (but the delivery time is only few days).

jp-001, 2018-05-10:
airwenn,

I hope you found a solution.
I am having the same issue you had. If I leave out DNS, gateway and netmask  on the Ethernet.begin call, the problem disappears.
With all parameters included, a client connection will seldomly succeed.
UDP traffic works, but with some errors.