ntruchsess/arduino_uip #175: How to change the CS pin in software?

petit-miner, 2017-06-14:
Hello, 
I would like to change the CS pin in software, because the CS pin is hardwired on a PCB.
How can I do that?

Bisotronic, 2017-09-28:
I think you can find something here (line #30)

**UIPEthernet\utility\Enc28J60Network.h**

`
#define ENC28J60_CONTROL_CS     SS

#define SPI_MOSI        MOSI

#define SPI_MISO        MISO

#define SPI_SCK         SCK

#define SPI_SS          SS
`

dimitar-kunchev, 2017-11-09:
I would like to expand on what Bistronic has suggested.

If you edit the header files for the library and you use it in different projects with different hardware setup you will run into trouble. Which is why the great guys of Arduino have made board definitions and board variants. If the CS is hard wired I guess then either the PCB is an Arduino shield (in which case it better use the default CS) or it is a custom board. I dealt with the latter case recently and after some googling I found it is actually quite easy to setup a custom board variant where you can specify what the SS pin is. Once you do that all the libraries will use the correct pins and work out of the box.

Google for boards.txt and board variants. The latter is where the magic of defining your custom board and its pin map happens.

petit-miner, 2017-11-24:
Nice, thank you. I will try that!

closed 2017-11-24 by petit-miner